uf-mil/mil #14: Compuer vision program to detect dock shape and color

kev-the-dev, 2019-09-18:
TODO: better description

See VRX rules re the docking task. We need a computer vision node to detect the color and shape of the elements on the dock. Given the ideal conditions of VRX, I feel confident this could be solved in a reliable way using traditional computer vision (OpenCV, edge detection, HSV color thresholding, contour and polygon approximations). 

We have some existing solutions for the real-life instance of this challenge (several in navigator_vision, including one using a ML model. These may be useful references, but the simulator will require at least different tuning or perhaps a simpler approach altogether.

kev-the-dev, 2019-11-04:
@JackGonzo11 @Cyang-UF updates? We're about three weeks away from the deadline. Even testing the existing solutions in VRX would be useful

JackGonzo11, 2019-11-06:
Spent the past week getting computer vision down. The only thing I have to do is get the implementation with ROS working with it. Shouldn’t take long(in theory) but you know how that goes.

closed 2019-12-11 by MarshallRawson