uf-mil/mil #116: SubjuGator odometry wrong in simulation

MarshallRawson, 2020-01-07:
On slower machines, when running the sub in gazebo, odometer gives the wrong position (transform from map) even when the simulation is paused.

It can be seen by running the simulation with gui:=true and then running the subviz.

kev-the-dev, 2020-01-09:
SubjuGator does not have any global knowledge of positions... our odometry is true odometry - measured relative to starting pose.

So this is a non issue UNLESS the roll/pitch angle is not aligned or our Z position is not an accurate measure of depth

MarshallRawson, 2020-01-09:
> SubjuGator does not have any global knowledge of positions... our odometry is true odometry - measured relative to starting pose.
> 
> So this is a non issue UNLESS the roll/pitch angle is not aligned or our Z position is not an accurate measure of depth

So technically, yes it is not strictly a solution breaking issue. However, I think it would be extremely useful for development if the transform between the Gazebo coordinate system and map was the identity transform no matter which machine is running the simulation.

kev-the-dev, 2020-01-09:
What is a use case of this functionality? I worry it would encourage developing things that rely on some global meaning to odometry that isn't actually present outside of sim. But first step is to see if the roll/pitch/z are problems

kev-the-dev, 2020-01-09:
Also my guess for a fix is to set a timeout delay for the sensor plugins to begin publishing so they don't start sending things into the kf until gazebo physics have stabilized after sub is spawned

MarshallRawson, 2020-01-09:
> What is a use case of this functionality? I worry it would encourage developing things that rely on some global meaning to odometry that isn't actually present outside of sim. But first step is to see if the roll/pitch/z are problems

So anything in perception that tries to estimate the location of an object would be extremely inconveinient to verify with this current bug.

Also, I have verified that the z looks correct, but roll, pitch,x, and y look incorrect.

MarshallRawson, 2020-01-09:
> Also my guess for a fix is to set a timeout delay for the sensor plugins to begin publishing so they don't start sending things into the kf until gazebo physics have stabilized after sub is spawned

Thanks I'll try that.

closed 2020-10-02 by MarshallRawson