uf-mil/mil #124: Fix / Document Net box

MarshallRawson, 2020-01-10:
The Network box is in a non-functional state. When plugged into SubjuGator, there is a yellow status LED on the ethernet connection, but no green LED indicating network connection. Also, the netbox needs to be documented so in the future if it needs to be fixed / replaced, we have a reference.

kev-the-dev, 2020-01-11:
With regards to the "fix" part... How do you know this is the net box issue not SubjuGator issue? Especially if your local machine or others can get a dhcp lease and connect to the router

MarshallRawson, 2020-01-11:
> With regards to the "fix" part... How do you know this is the net box issue not SubjuGator issue? Especially if your local machine or others can get a dhcp lease and connect to the router

Using a different router, the sub was giving the green LED and visible on network, that is how I got the competition changes on [SubjuGator competition branch](https://github.com/uf-mil/SubjuGator/tree/competition)

kev-the-dev, 2020-01-12:
Confirmed cannot repro, this is a documentation ticket

PR #126 (2020-01-15): Document network box
Fixes #124

This turned into a monolith PR as many things build on each other. Main features:
* Add docs page for network box
* Add docs page for branding
* Some other reorganizing of docs
* Add docs / scripts for connecting to MIL file server
* God I love docs
closed 2020-01-15 by kev-the-dev