uf-mil/mil #135: C++ Mission System

kev-the-dev, 2020-01-23:
From @RustyBamboo 's suggestion, it may be time to ditch our Python-based mission system. This system was designed for quick development and on the fly changes, but a mission system should really have different properties. If behavior needs to be tuned / configured on the fly, we should use config files, parameters, etc, not changing script code. The time loss in developing in a systems language is justified by avoiding syntax errors, type errors, and other cases that are embarrassing and costly at field testing / competition and which compiled languages mitigates.

Here's my proposal:
* Maintain API compatibility with the existing mil_missions package. Essentially, this will be a C++ rewrite of mil_missions. This allows our client-side tooling, including the mission GUI, to continue to work with the old and new system
* To maintain similar functionality to the txros system (allowing you to write async ROS code in a linear manner, such as waiting for actions or messages), implement a roscpp wrapper written using `std::async` and `std::futures`.  More needs to be designed here, but simply using roscpp would probably not make for readable mission code
* Focus on re-usability across MIL platforms. e.g. the C++ version of PoseEditor should only need to be written once, same for abstractions like computer vision interface
* ??? Make use of pluginlib to allow missions to be compiled as separate targets and reloaded on the fly, potentially without restarting the mission server
* Use SubjuGator as the first test platform for this system

For funsies, pinging @mattlangford @jpanikulam who might have some ideas on how to do this right / if this is a good idea at all

RustyBamboo, 2020-01-27:
This is great. I think we need to rethink how we approach designing missions in the first place. At a high level, we should be able describe what the mission/missions should do, what resources they take and give, as well as how they work together. To model this, we could use [petri nets](https://en.wikipedia.org/wiki/Petri_net). This would give a nice description of how the missions should work and how they interact, in an asynchronous manner, while also supporting non-deterministic scenarios. 

From there, C++ rewrite should respect the Petri model (async support, mutex, etc). And likewise, all the written missions will follow the framework, hence respecting the model. 

So what I envision is an initial petri net drawing before competition, followed by delegation of subgraphs to programmers. Because of the resource planned out, the mission system we can easily integrate multiple missions and seamlessly support concurrent missions.

Simulating might be a little tricker, but seems like there are C++ libraries to simulate Petri nets.

kev-the-dev, 2020-01-29:
This sounds like an interesting model for doing missions - my worries are implementing this in a way such that inexperienced developers can quickly contribute mission code. Do you want to be assigned to this project, as you have such a vision for how it should be done?

MarshallRawson, 2020-09-23:
See #267 for a working core petri net based system!