uf-mil/mil #143: Hydrophone Driver

MarshallRawson, 2020-01-26:
Tweak the Sylphase passive sonar driver to be able to detect the direction of one the frequency (with a margin) we are interested in, not simply the loudest frequency we hear.

kev-the-dev, 2020-01-27:
Step 1 is actually get the current version of this software available to whoever wants to work on it

kev-the-dev, 2020-01-27:
For my reference, this software is hosted on a private gitlab https://gitlab.sylphase.com/sylphase/sonar, and @RustyBamboo made some changes to it locally which should be present in the SubjuGator backup.

I do not know which version we ran with at competition or what changes were made.

kev-the-dev, 2020-01-27:
My personal preference would be to ditch the proprietary package as much as possible - use it is a driver to stream raw samples to a ROS node that WE develop within the monorepo for processing / filtering.

kev-the-dev, 2020-01-29:
I messaged Forrest and he suggusted the same thing - set the API boundry at the raw tcp stream of samples and he gave permission to open source all the processing that happens on the other end of that API, including some graphing scripts. This is done in #146.

I would still really like the raw samples to be in a ROS message so that they can be bagged synchronously with our other data, but we'll have to figure out how to do that

closed 2020-04-08 by MarshallRawson