uf-mil/mil #166: Investigate collision of flow for Thrusters

jaxnb, 2020-02-03:
Thrusters currently have overlapping flow paths. Does this matter? If so, figure out a way to angle the thrusters to deal with that.

MechAnger, 2020-02-06:
If the thrusters jet flow will intercept with other parts, the references show that there will be up to 46 percent loss.  If placed at an angle between 30-40 deg(to offset the jet flow interception, also by other factors of the subs spine know as Coanda effect), the obvious downside is decreased forward thrust by couple of percent , but the upside is increased angular torque(yaw). Also I have the CFD from Solidworks simulation included to show the flow but the references use better software's. The other thing that the references point out is to put the thrusters under approximately 6-8 deg pointing down(away from the spine), these losses can be minimized even more. I believe extending the sub(which is also needed for the main shoe box convection issues) is necessary to offset the hor. thrusters jet flow interception from spine and vertical thrusters behind them.  I included the links in the thrusters grab cad folder

andrespulido8, 2020-02-26:
Thrusters changed location to minimize flow interference

closed 2020-02-26 by andrespulido8