uf-mil/mil #170: Solidworks calculate buoyancy or sealed volume

jaxnb, 2020-02-03:
Would be nice to find an easy way to calculate the volume of the sub when sealed. Buoyancy numbers can help us dial in the buoyancy of the sub.

MechAnger, 2020-02-05:
I started to look into this problem. Unfortunately, Solidworks does not have the buoyancy feature or simulation. The only way is to recreate the same assembly that we have with a different configuration with no hollow pieces and change that configuration's material to water. Measure the mass of the original configuration and make the 2nd configuration's mass with water same since (CENTER OF BUOYANCY IS THE CENTER OF DISPLACED VOLUME). Then measure CG of the original configuration and the water's configuration (assuming mass was set the same in both materials). See the difference.

Reference:
https://forum.solidworks.com/thread/73001

jaxnb, 2020-02-06:
Maybe its worth checking if 3rd party software can do it?

closed 2021-01-29 by MechAnger