uf-mil/mil #179: Machine and mount new buoyancy foam

andrespulido8, 2020-02-09:
Keep same dimension from old foam. Paint them black. Calculate the buoyancy force per unit of volume for the foam so we know the force it is required to maintain the sub slightly positively buoyant

andrespulido8, 2020-02-09:
Sub right now is negatively buoyant when Imaging Sonar is mounted

jaxnb, 2020-02-10:
What did we add/remove? I'd say that sub being negatively buoyant in the same state as it was at the competition would be a concern. We should correct this, but also pay close attention to what has changed.

closed 2021-02-04 by MechAnger

andrespulido8, 2021-02-04:
Is the new foam purchased already machined and mounted? @MechAnger

MechAnger, 2021-02-04:
i opened another issue while there was a physical meeting and did  not see this one, i have copied all the comments under "Blue Robotics R-3312 Foam Buoyancy Measure and Cut #284"