uf-mil/mil #205: INDY AV: Waypoint Saving Node

popeyedpanther, 2020-03-12:
To create a waypoint following software stack we need a node that:
Subscribes:
 - /odom

Publishes:
 -none/TBD

Functionality:
The node should subscribe to the /odom topic and at some predefined rate (maybe a param or arg) and save the data to a file, preferably a yaml file. 

The saved data should be at the minimum:
 - Position (2D or 3D?) in meters
 - Velocity (Speed or linear velocity vector?) in meters/second

MarshallRawson, 2020-03-12:
I think this node should also save relevent details of the saved path. ie: ECEF of start, date created, and a name(passed in as a param)

popeyedpanther, 2020-03-12:
As a side note, where should this code go in the repository, It doesn't really fit into the directories I have set up so far. Its more of a utility than a critical part of the waypoint following autonomy.

JoshuaSiy, 2020-03-12:
How do i assign myself?

popeyedpanther, 2020-03-12:
I assigned you but on the top right it should say "Assignees", just click the gear and find your account in the list.

closed 2020-04-08 by MarshallRawson