uf-mil/mil #232: Look into replacements for regulatos

andrespulido8, 2020-06-04:
The regulators for the paintball tank are not available to buy anymore. A need regulator for the paintball tank needs to be bought, or maybe replace all the paintball tanks with a dive tank. needs to buy. Additionally, look for a safety valve with higher pressure threshold for the pneumatic manifold (150psi).

MechAnger, 2021-06-28:
We will introduce servos from sub 9 and get rid of the valve box and paintball tank. So whoever designs the torpedo launcher and gripper for sub 9 using servos, please also do it for sub 8!

andrespulido8, 2021-07-01:
Before saying we're getting rid of pneumatics have all the electric actuators manufactured and make sure they are as or more reliable than the pneumatics

closed 2024-04-14 by uf-mil-bot