uf-mil/mil #240: Finish implimenting the Sub9 model

MarshallRawson, 2020-08-10:
There is currently a bug with sub9 urdf that causes the odometry_estimator node to spit out nothing but nans. checkout `sub9-model` branch to see what I am talking about

colefisher1, 2020-08-12:
After reading through your commit and testing sub9 locally, I was able to pinpoint the exact line causing the error to line 20 in the sub9.urdf:
`<inertia ixx="0.4674" ixy="-0.003" ixz="0.02" iyy="0.983" iyz="-0.0003" izz="1.197" />`

I found that removing or commenting out this line from the urdf fixed the nan issue. The source of the error is almost definitely one or multiple data values in the inertia matrix. 

See the link below:
 https://answers.ros.org/question/224380/nan-value-after-changing-inertia-matrix/

MechAnger, 2020-08-14:
Found an exporter from direct solidworks to URDF that creates all the configs and launch files with ability to edit the origin, inertia matrices, meshed,textures and more that might solve this issue in the future since we do not have to input those infos anymore and it will be automatically generated. Already tried with other assemblies and it seems to be working in gazebo and rviz.

http://wiki.ros.org/sw_urdf_exporter

cbrxyz, 2024-04-14:
Fixed

closed 2024-04-14 by cbrxyz