uf-mil/mil #248: Investigate/Design a new electrical torpedo launcher

andrespulido8, 2020-09-03:
Since the motivation is not to use pneumatic for the new sub. An investigation about new torpedo launcher needs to be conducted. Look for the design of how other teams have approached this problem (find their videos, papers, and websites for more info) , or design one from scratch.
Website: https://robosub.org/programs/2019/.
Contact: Andres Pulido

MechAnger, 2020-09-14:
Ruben Gonzalez-Vera is assigned

andrespulido8, 2022-01-10:
@MihirRao01 what is the path to the torpedo design in GrabCAD?

MihirRao01, 2022-01-11:
I believe it is subjuGator 9 Cad ->Electric Torpedo Launcher @andrespulido8

andrespulido8, 2022-01-11:
@MihirRao01 mount to the sub appears to be missing in the assembly. Also lets manufacture (print) it before closing for sure

MihirRao01, 2022-01-11:
@andrespulido8 I closed it because there was another issue open focusing on the torpedo launcher. I wanted to get rid of any duplicates to avoid confusion.

MechAnger, 2022-01-11:
#351

closed 2022-01-11 by andrespulido8