uf-mil/mil #252: Write autozeroing into the CNC

andrespulido8, 2020-09-11:
Add the capability to auto zero the CNC. You will learn how to implement it using G code.
Resource: Daniel Pestridge - He loves substractive manufacturing
                 Jaxon Brown: he donated the CNC

jaxnb, 2020-09-11:
The CNC should already support automatic x and z zeroing, with y zeroing being available through the automatic Conductivity zeroer. What are we trying to accomplish that these tools don’t solve?

andrespulido8, 2020-09-12:
> The CNC should already support automatic x and z zeroing, with y zeroing being available through the automatic Conductivity zeroer. What are we trying to accomplish that these tools don’t solve?
"Yeah the it would be a bitsetter capability between toolchanges. Now it can autozero to the workpiece but I dont see a toolsetter capability macro unless I'm missing something. The purpose is to get relative heights between tool changes that it could run. There are macros but im unaware of the ones on the machine for that if they exist." Dan

andrespulido8, 2020-12-28:
Dan comments: The pendent has been problematic and so has the z height probe but they can be used just be wary. I would look at adding a script to a tool setter if possible so that you can do tool changes and keep runs going. This might be easy or not I think there are prewritten macros on the internet but making them is t that bad. Otherwise, one tool at a time is the most reliable I've seen.

closed 2024-04-14 by uf-mil-bot