uf-mil/mil #255: Design a new hydrohpone housing for Sub8

andrespulido8, 2020-09-13:
We currently house the hydrophones and its box inside a housing made for them. The goal is to move the actual hydrophone sensors as far as possible to any air  containers because we learned during the competition that even small bubbles can affect the sensors. The ideas to solve it are to either house the board inside the main PV (pressure vessel) or have a box made only for the board with the sensors far away attached to preferably metal rods. Talk to an electrical team member to check feasibly in terms of cabling.

Contact: Andres Pulido

MechAnger, 2020-09-14:
Jose Lora is assigned

MechAnger, 2021-06-17:
Increase the volume in the enclosure, get rid of the clamp collar axle with set screw for phones and come up with better solution, add one more groove (double boring on the seal of the enclosing)(only has 1 now), move down, and come up with better mounting to the plexiglass as well (linear railing is great).

MechAnger, 2022-04-20:
@MihirRao01

closed 2024-04-14 by uf-mil-bot