uf-mil/mil #271: Gantt chart plugin for the mil github

MarshallRawson, 2020-10-04:
find / write a plugin to link the MIL issues to some Gantt chart infrastructure

sferekides, 2020-10-05:
Monday.com

- [X] Task dependent on those that need to finish first

- [X] Start and end date on Gantt chart

- [X] Task cannot be scheduled to a start date when dependencies are not done

- [X] Visualization of Gantt

- [ ] Viewable on linux (tbd)

- [ ] Divide tasks into one or more categories and are visually independent

- [ ] Issue opened - task appears (have to manually set date or have automation pick a date)

- [ ] when issue is closed (have not found way to automatically mark it as closed)

- [ ] source exists in mil repo?

MarshallRawson, 2020-10-05:
> 
> 
> Monday.com
> 
>     * [x]  Task dependent on those that need to finish first
> 
>     * [x]  Start and end date on Gantt chart
> 
>     * [x]  Task cannot be scheduled to a start date when dependencies are not done
> 
>     * [x]  Visualization of Gantt
> 
>     * [ ]  Viewable on linux (tbd)
> 
>     * [x]  Divide tasks into one or more categories and are visually independent
> 
>     * [x]  Issue opened - task appears (have to manually set date or have automation pick a date)
> 
>     * [ ]  when issue is closed (have not found way to automatically mark it as closed)
> 
>     * [ ]  source exists in mil repo?
> 
> 
> A lot of cool automations for the table/Gantt chart where you can set up dependencies and status. Big issue is that when you close an issue on Github it does not have an integration that automatically marks it as closed in the Gantt.

is Monday.com open source / free?

sferekides, 2020-10-05:
My initial impression is that it is free up to a certain amount of automated actions per month. You get 25,000 free automations each month. I am going to look into pricing more.

sferekides, 2020-10-08:
OpenProject
- [x]  Task dependent on those that need to finish first

- [x] Start and end date on Gantt chart

- [ ] Task cannot be scheduled to a start date when dependencies are not done ( I believe so but hard to tell for sure)

- [x] Visualization of Gantt

- [x] Viewable on linux

- [x] Divide tasks into one or more categories and are visually independent

- [ ] Issue opened - task appears - can use github plugin but did  not find one that I was sure would work for us

- [ ] when issue is closed - can use github plugin but did not find one that I was sure would work for us

- [x] source exists in mil repo - it could exist in source but we would also need a PostgreSQL database => I don't think locally hosting the database makes sense

- [x] Free! (if we don't want their cloud hosting)

TLDR: free and cool resource; biggest issue is having to host PostgresSQL database, but we could have OpenProject in the mil repo; another issue is that it is not clear that github would integrate well; will likely have to look for plugins/make plugin that supports a github integration that fits what we want

sferekides, 2020-10-08:
Redmine
 

- [x] Task dependent on those that need to finish first

- [x] Start and end date on Gantt chart

- [x] Task cannot be scheduled to a start date when dependencies are not done (don't know for sure)

- [x] Visualization of Gantt

- [x] Viewable on linux

- [x] Divide tasks into one or more categories and are visually independent

- [x] Issue opened - task appears - Zapier integration available

- [ ] when issue is closed - Zapier integration available (limit on free account) - I can't find anything about closing a task in Redmine when the issue closes in github;

- [x] source exists in mil repo - it could exist in source but we would also need a database that we would have to host (more options for database types than OpenProject

- [x] Free! 

TLDR: similar to OpenProject where we would need hosting. Plugins seem more intuitive for github integration but also have some zapier functionality that could be a solution. Again we have the issue on github functionality is not certain.

sferekides, 2020-10-08:
I can look at more alternative after we discuss this tonight at the meeting.

sferekides, 2020-10-08:
To do: figure out how hard it is to write a plugin/set up database/ host on MIL network

cbrxyz, 2024-04-14:
No longer being implemented (we use GitHub Projects now)

closed 2024-04-14 by cbrxyz