uf-mil/mil #276: Research into supervised machine learning on colored meshses

MarshallRawson, 2020-10-08:
In MIL, we often deal with multiple sensor inputs on different robots taking in different data from different sensors, however, they can almost always be reduced down to a colored mesh. That is: some number of colored triangles in 3d space that represent the surface of the object in question. If we could make a generic machine learning framework around this data type, we could have a single elegant solution for most robot systems.

MarshallRawson, 2020-10-08:
ideally the framework would work like this:
inputs:
* models of the known elements to look out for
* data about the world
 -> expectation maximization ->
outputs:
* the locations and orientations of the elements that is has found

note: there is no garuntee how many of each element is in the world

andrew-aj, 2020-10-08:
Here are some research papers that I found which are related to our task:
http://robots.stanford.edu/papers/thrun.3dmap.pdf
https://smartech.gatech.edu/bitstream/handle/1853/42364/05652091.pdf

vishwa-11, 2020-10-23:
The Expectation Maximization (EM) algorithm is an iterative statistical method in unsupervised learning used to estimate latent variables, given a set of incomplete, observed data. Latent variables are variables that we know exist but are not explicitly observed and directly affect the value of observed data points. For example, we may know a person's 100m sprint timing (observed data) but not his age (latent variable). However, these two are very much related. 
The EM algorithm works by taking a random estimate of a latent variable, and then calculating the expectation of the probability of the observed events given this random estimated. Then, a new variable for the latent variable is chosen to maximize the probability calculated in the previous step. This process is iterated until both the expected and maximized probabilities are similar or equal. This results in an estimated value of the latent parameter.

closed 2022-05-05 by andrew-aj