uf-mil/mil #282: Upgrade the CNC set up (many tasks)

andrespulido8, 2020-12-29:
If you can do one of these, write it in the comments. 

1. - The vacuum setup currently in use works fine but it is limited the y movement. The previous solution by Daniel Prestrisdge was to purchase a dust boot from carbide 3d and shaved off a bit of the inside of the lip to fit. He suggests to get two sets and replace the brushes with 1" and 2" from McMaster. 
2. - Get a fogbuster or equivalent so that there are fine adjustability and predictable misting/air blast. Suggest getting a hard middle on two sides to get great coverage for better cuts. The single nozzle can have "blind spots".
3. - Update the computer to the newest inventor/SW/fusion or so the cuts can be as seamless as possible from designs.
4. - Add impellers to the coolant line to ensure visual flow. (https://www.amazon.com/Alphacool-Eisfluegel-Flow-Indicator-Clear/dp/B01HQ8JT0K/ref=mp_s_a_1_2?dchild=1&keywords=cnc+flow+indicator&qid=1609184939&sr=8-2)

MechAnger, 2021-01-21:
5) •The rails need to be lubed (hg50 linear bearings I think)  and may even have provisions for internal lube like a speed fitting in place of info the screw but either way this should be conducted to preserve the life of the rails.
6) •The water in the bucket was slimy. Replacing with a newer lore rigid tubing would be helpful.
7) •There is no dust from the mdf which can be devastating in terms of cleaning up and health wise so the suggestion switching over to hdpe or aluminum and just surfacing that.Bad work holding can be detrimental to your work-piece and minimizing vibrations is key.
8) •The pendent has been problematic and so has the z height probe but they can be used just be wary. Look at adding a script to a tool setter if possible so that you can do tool changes and keep runs going. 
9) •Buy tape and glue. I recommend 2p10 glue and accelerator. Frog tape but I've hear some carpet tape is good.
10) •Keep plenty of isopropyl on hand for aluminum coolant and maybe consider a different container

MechAnger, 2021-01-28:
11. • Use or purchase new stock for the enclosure for CNC machine against the small pieces flying all over the place and add some sort of plexiglass for visibility during cut at the same time for safety for the user.

MechAnger, 2021-01-28:
Garett McKee is assigned

MechAnger, 2021-02-02:
12) Need to add to the water is some kind of bacteria killer since it's not a closed system. Most people use a specific spindle coolant or RV antifreeze from Walmart or anywhere else, or plain bacteria acid type.

MechAnger, 2021-02-04:
Documentation will be needed as well for these issues if we ever run in the future into them!

andrespulido8, 2021-02-04:
I can teach a mech in charge of this how to do documentation in our site @MechAnger

MechAnger, 2021-02-04:
Perfect! We will tape everything and have one of us make a documentation. It is crucial to whoever makes it, be with us whilst we fix the CNC.

MechAnger, 2021-02-17:
Everything is completed except an order of HDPE needs to be done, water lines need to be changed, dust boot from sweepy needs to be ordered and replaced (may need to be shaved off a little for perfect fitting when the z-axis moves)

MechAnger, 2021-03-05:
Everything is complete except the dust boot, replacing the water lines and adding the flow meter with HDPE.

closed 2024-04-14 by uf-mil-bot