uf-mil/mil #298: Design mount for Hall Effect Sensor

andrespulido8, 2021-03-28:
The hall Effect sensor (a magnet taped into the main vessel frame) that controls the hardware kill needs a 3D printed mount to prevent it from coming loose.

Talk to an electrical to test that mount is working

MechAnger, 2021-06-17:
Also, design the new kill wand (with a magnet cutout) and a way to permanently mount the kill wand onto the main vessel outside without velcro. PLA and ABS is suggested for simplicity for both.

MechAnger, 2021-06-28:
[image]

MechAnger, 2021-06-28:
similar to this where the pvc pipe is currently at on sub 8

cbrxyz, 2022-09-19:
The Hall Effect Sensor on SubjuGator has now been deprecated in favor of an electrical kill system.

closed 2022-09-19 by cbrxyz