uf-mil/mil #299: Fixing the wiring in the Trailer.

kawaiiPlat, 2021-04-06:
The wiring in the trailer that powers the AC, lights, and power outlets is in some state of disrepair or midway through a project. We can power the laptops and monitors by running cables to them from the generators currently, but the wiring needs to be repaired eventually. As best as we can tell, someone was trying to wire up a UPS to the breaker box inside the cabinet. When you connect the outside cables to the generators, the power system inside will hiccup on and off, and it even caused the generators to turn off. It seems we have a short somewhere...

TLDR We need to get a proper (read housing and code knowledgeable) electrician to check it out and figure out whats up.

MechAnger, 2021-11-22:
The entire system is also wired backwards ( all the male plugs have 120V running though them) per Alex K.

ems37uf, 2021-11-22:
I sent the following to Daniel Preston (MAEC shipping) <prestond@ufl.edu> earlier today.

Hello, Dan. Dr. Crane told that you are the one to contact about some work that we need done on the CIMAR/MAE trailer parked on the south side of MAEB. (Similar issues previously were handled by Jeff Studstill. In fact, Jeff started some refurbishing of the trailer before he left, but the below updates were never completed.)

The wiring in the trailer that powers the AC, lights, and power outlets is in some state of disrepair or midway through a project. We can power the laptops and monitors by running cables to them from the generators  currently, but the wiring needs to be repaired. As best as we can tell, someone was trying to rewire our UPS to the breaker box inside the cabinet (near the front of the trailer). When you connect the outside cables to the generators, the power system inside will hiccup on and off, and it even caused the generators to turn off. It seems we have a short somewhere. Also, the entire system is also wired backwards (all the male plugs have 120V running though them), which is definitely a huge electrical hazard.

We have two generators presently stored inside the trailer which we use when we are doing field testing.

Any info that you can provide to get the repair process started would be greatly appreciated.

kawaiiPlat, 2022-05-20:
Trailer wiring has been repaired, external GFCI replaced, lights checked. Generators need basic maintenance and a operation guide for new students. All appears stable.

closed 2022-05-20 by kawaiiPlat