uf-mil/mil #321: 3D print a downsized model of the sub 9 frame

castry434, 2021-09-03:
Before we can send the drawing for sub 9's frame to a workshop so they can use a laser cutter to cut the aluminum, Dr. Schwartz wants us to ensure that the parts of the frame will fit properly by 3D printing a scaled-down version of the sub frame. For this task, you should create a copy of the frame's assembly rather than modifying the whole assembly for the submarine.

(Dr. Schwartz would like to have this 3D printed model in his office so try to make it high quality.)

Contact Andres Castrillon on Slack before getting started with this task.

J4mZzy, 2021-09-10:
I will be working on this issue

MechAnger, 2021-11-27:
The structure was given to Dr.Schwartz.

closed 2021-11-27 by MechAnger