uf-mil/mil #322: Determine why NaviGator wrench/autonomous skyrockets when wayfinding

alexperez33, 2021-09-05:
On master branch, run ```roslaunch navigator_launch vrx.launch --screen```.
Then on one pane run ```vrxviz``` and one another run ```runmission Move f 5```
This is the proper behavior the vehicle is supposed to have.

However, when this is run in the vrx_2022 branch, the boat goes really far, really fast, almost ignoring its goal. Its force values can be seen if you do ```rostopic echo /wrench/autonomous```.
This topic is published by the mrac node, which can be found by saying:
```roscd navigator_controller/nodes```
**The task is to figure out why the mrac is publishing these weird wrench values.**

To get to the vrx_2022 branch.
```git checkout vrx_2022 && git submodule update --init --recursive```
Make sure to run these commands as well or the code won't properly run.
```cd ~/catkin_ws/src/mil/NaviGator/simulation/VRX/vrx && git checkout master```
```cd ~/catkin_ws && rm -rf build devel && cm```

If you would like to make changes, create a new branch off of the vrx_2022, and push your changes to your new branch.

If you want to go back to the master branch of the repo, make sure you have no changes or have them committed:
```git checkout master && git submodule update --init --recursive```
```cd ~/catkin_ws && rm -rf build devel && cm```

NOTE: Recently I have attempted to incorporate the new VRX code (https://github.com/osrf/vrx) with our current NaviGator code. One major change they did was the addition of the "wamv/" namespace on all the urdf links. This can be seen when you have the vrx.launch file running and run ```rosrun tf view_frames``` and after that finishes run ```evince frames.pdf```.
This will show you the structure of the boat and how it is all connected. (make sure to delete the .pdf and .gv file it produces when done). Regardless, this is something you can run on the master branch as well and see how it is different.

Due to this, I made some changes to get the simulation to work again. I am unsure if these changes are the cause of the problem, or if there is something that still needs to be considered, but it is something to be aware of. The changes I made can be seen by running ```git diff master vrx_2022```.

IMPORTANT: Changes can only be made to our NaviGator code. anything within the ~/catkin_ws/src/mil/NaviGator/simulation/VRX/vrx directory cannot be changed.

If you have any questions, please contact Alex Perez in slack.

alexperez33, 2021-10-02:
Vrx has resolved this issue. Issue was not fully related to our code.

closed 2021-10-02 by alexperez33

andrespulido8, 2021-10-03:
nice