uf-mil/mil #340: Export components from Sub 9 assembly as STL

castry434, 2021-09-21:
Any components in the Sub 9 assembly that are ready to be 3D printed need to be exported as STL files so I can send them to the rapid prototyping lab for printing. Along with the STL file we also need to know the maximum dimension for each part so the lab can scale our print properly. This dimension can be found using the "evaluate" feature on SolidWorks.

Create a new file inside of the SubjuGator 9 folder called "Prints Fall 2021". This is where you will save all the STL files along with a screenshot of the maximum dimension. For each component, make a folder with those two files (the stl and the image) so we can keep all of the files organized and next to each other.

Some of the components to export include:
-Sub 9 feet 
-Vertical Thruster mount
-Uclamp
-Support for acoustic modem and it's clamp
-Carbon tube bottom clamp
-Carbon tube top clamp
-Imaging sonar base
-Imaging sonar top
-Imaging sonar bottom

Go through the assembly to make sure there aren't any others. Also, make sure to take note of how many copies will need to be printed for each files so we can order the correct number.

Contact Andres Castrillon on Slack if you have any questions.

MechAnger, 2021-09-21:
the support of the acoustic modem and the clamp was printed and is on the table where sub9 will be. The tube was purchased and delivered as well as endcaps.

cbrxyz, 2024-07-17:
Mechanical issue

closed 2024-07-17 by cbrxyz