uf-mil/mil #494: Tasks Building for SubjuGator 2022

MechAnger, 2022-03-22:
Someone needs to physically to start building tasks for robosub competition for summer 2022. Drawings are inside this PDF. PVC pipes are inside or outside in the back of CIMAR (by the trailer). PVC cutter is in the main lab drawer where all the cutting tools are. Pictures can be put on a poster and lamination of the pictures and the pictures print outs can be done in target copy right across UF. Wood planks can be found in mail lab or CIMAR. Can use band saw to cut it to shape as it asks. Star Gate needs to be build first (last page of the PDF) as it will serve as the qualification video that we will need to submit. Also good to start on other tasks before may/june so more testing can be done.
[RoboSub-2022-Task-Ideas.pdf](https://github.com/uf-mil/mil/files/8327579/RoboSub-2022-Task-Ideas.pdf)
.

kawaiiPlat, 2022-06-12:
Elements that need to be manufactured (Section of the Handbook)
- [ ] ***Start gate (2.5.2)***
  - [ ] Current build is not the correct dimensions or color
- [ ]  ***Path Marker (2.5.3)***
- [ ] ***Bouys (2.5.4)***
  - [ ] One of the images is already printed
- [ ] ***Bins (2.5.5)***
- [ ] ***Shootout (2.5.6)***
- [ ] ***Octagon (2.5.7)***
  - [ ] Octagon
  - [ ] G-Man table
  - [ ] Bootlegger table
  - [ ] Bottle Table

cbrxyz, 2022-07-25:
Duplicate of #608

closed 2022-07-25 by cbrxyz