uf-mil/mil #596: Get passive sonar working

alexperez33, 2022-05-14:
On Sub8, we have a passive sonar board that has hydrophones connected to it. These hydrophones listen to an acoustic beacon and calculate the position and direction of the beacon. We need to ensure the hydrophones are still intact and working, and that the board is still functional. I have documentation related to this if you are interested in learning how the passive sonar works and ensuring that is does work throughout the semester.

This task is dependent on the new passive sonar enclosure working. We need to ensure the device that is supposed to emit the frequency (the beacon) works properly, and we need to go out and test that this works in the new enclosure. If it doesn't work, we need to debug what is going wrong. This is a crucial component to the competition.

kawaiiPlat, 2022-06-05:
Removed hydrophones from old enclosure. Tested them with the new sensor board using the basic graphing testing scripts. Also tested them in a small container of water, and when placed in the water they seemed to be triggering on the correct frequency. We weren't able to get a position estimate for the pinger, but the water body was rather small. Will try to test the hydrophones in water during the next sub test day.

kawaiiPlat, 2022-08-01:
@mehront , @aramesh10 did various modifications during the 2022 robosub competition. Changes include shortening the cables on the hydrophones, replacing the weird double-sided usb micro B with a standard micro B, and getting more info on reliability from Forrest. Info to come in the documentation.

kawaiiPlat, 2022-08-29:
This issue wasn't specifically resolved, more details are in #774

PR #774 (2022-08-09): Lessons from RoboSub
Adding documentation on lessons we've learned from RoboSub 2022.

Resolves #596
closed 2022-08-29 by kawaiiPlat