uf-mil/mil #597: Investigate PNP

alexperez33, 2022-05-14:
Look into PNP (https://en.wikipedia.org/wiki/Perspective-n-Point) and sift through code to see how we have implemented them in the past. Determine if it will be useful for our current tasks for RoboSub and RobotX.

cbrxyz, 2022-05-14:
Assigned myself to this task because I am attempting to understand `mil_vision` so that the software and methodologies behind it can be documented, as part of the Noetic + Alabaster project. (#459)

Because this task is not attached to any specific action, and rather just a general topic, someone else can hop on here alongside me if you'd like.

cbrxyz, 2022-09-30:
Updating the task to relate specifically to adding documentation.