uf-mil/mil #599: Get hyper spectral camera working

alexperez33, 2022-05-14:
We currently have a hyper spectral camera for the RobotX competition. This will be placed on the drone to allow us to see certain signatures of various wavelengths. This information will be relayed to the boat and allow the boat to complete the corresponding task. However, the first step is to ensure the camera works in the first place. Soon we should be receiving the proper POE injector to turn on the camera. If it turns on, there is code online from OpenHSI that helps with using the camera. Also there is another link to more information on how to debug the camera.

https://openhsi.github.io/
https://robonationforum.vbulletin.net/forum/robotx/-2022-robotx-challenge/3834-using-the-hyperspectral-camera

Once we ensure it works and we can get data from it, we will work on integrating the camera into our sensor suite.

mitchellrobert, 2022-06-12:
Here is a link to the reflectance profiles that we will have to detect for the wildlife challenge: https://robotx.org/programs/robotx-challenge-2022/#resources
This is a link to a technical document about the camera: https://www.dropbox.com/s/3akr8s9h111yi2i/OpenHSI_Tech_summary.pdf?dl=0
And this is a link to the calibration files for our camera of serial number OpenHSI-18: https://www.dropbox.com/sh/77lrkw3e6ntburh/AABz9hlOI3kNcSz_vhOSi5TFa?dl=0

cbrxyz, 2022-09-23:
This issue has been superseded by #818. Thanks for your help with this Mitchell!

closed 2022-09-23 by cbrxyz