uf-mil/mil #601: Investigate current heartbeat system for RobotX and determine what needs to change

alexperez33, 2022-05-14:
Currently there is a package called ```navigator_robotx_comms```. This package deals with communications between the boat and the land and reports the status of the boat at any moment in time. We need to determine how this system works, why it works like this (based on the 2019 task description), and what we need to change based on the (2022 task description).

https://robonation.org/app/uploads/sites/2/2019/09/RobotX-2018-Tasks-FINAL_v4.0.pdf
https://robonation.org/app/uploads/sites/2/2021/06/2022-Maritime-RobotX-Task-Descriptions-and-Specifications_v1.0.pdf

MyleeSmith, 2022-07-22:
HEARTBEAT SPECS 2019

Situational Awareness Reporting Task (2019)
Situational awareness is an integral component of unmanned systems. Understanding an unmanned system’s intent is critical in building operator trust. The 2018 RobotX Communications Protocol (see Appendix C) provides a standardized protocol through which heartbeat and task information may be communicated to judges and TD staff.

Network Information (2019)
During operations, teams will be provided with a hard-wired connection (RJ-45) to the Technical Director’s network. This connection must be used to transmit the Autonomous Maritime System (AMS) heartbeat and other reports. When connected to the TD network, the team’s computer must request an IP address from a TD Network DHCP server. Once connected, they should establish a TCP connection to a server with an address and port number, correlating to the selected course. Address and port numbers for each course will be provided during the event. A unique NMEA sentence has been defined for each challenge requiring communication between the AMS and a judge.

SPECS 2019
[image]

HEARTBEAT SPECS 2022

System Heartbeat (2022)
In previous RobotX Challenges (2016, and 2018) teams were required to implement a visual feedback system and a heartbeat broadcast system.
This will continue to be part of the 2022 RobotX Challenge requirements. Teams at each of the course operations tents will be provided with a wired RJ45 connection. Information from the team’s Operator Control Station (OCS) will be transmitted to the Technical Director (TD) network, using this wired connection. Teams are expected to provide their own wireless link for information exchange between the AMS and their OCS. Details regarding communications protocol can be found in Appendix C: 2022 RobotX Communications Protocol.

Network Information (2022)
During operations, teams will be provided with a hard-wired connection (RJ-45) to the
Technical Director’s network. This connection must be used to transmit the Autonomous
Maritime System (AMS) heartbeat and other reports.

SPECS 2022
[image]

CHANGES:
* Change from AUV to UAV
  * Add faulted state
* Change date and time zone (no longer Hawaiian time + change from HST to AEDT)

uf-mil-investigator, 2022-07-22:
@MyleeSmith, hola from the [Machine Intelligence Lab at the University of Florida](https://mil.ufl.edu)!

  Unfortunately, at this time, we are unable to accept issues and contributions to this repository from users who are not yet part of MIL. We may accept contributions in the future, and some of our other repositories and projects accept contributions. However, because much of our lab is dependent on this repository and because we use several in-house tools to ensure that new additions work properly, we are unable to easily accept the contributions from researchers outside of MIL.

  _If you are a member of MIL, please contact a software leader to get access to the @uf-mil organization._

kawaiiPlat, 2022-07-23:
investigation seems complete, closing. Good job @GraceBorst and @MyleeSmith

cbrxyz, 2022-07-24:
I think with this task, it might be worth keeping it open a little longer since the code needs to change in order to accommodate the 2022 season. The task title should probably be adjusted to suggest this.

closed 2024-03-25 by cbrxyz