uf-mil/mil #620: Docs: Dead link to old development guide.

kawaiiPlat, 2022-05-26:
The main [SubjuGator](https://mil.readthedocs.io/en/latest/docs/subjugator/index.html) page in the docs has a link to the (now deprecated) [Development Guide](https://mil.readthedocs.io/en/latest/docs/development/development_guide.html), this should probably point to [Getting Started](https://mil.readthedocs.io/en/latest/docs/development/getting_started.html) instead.

[This](https://mil.readthedocs.io/en/latest/docs/development/adding_documentation.html) gives a guide on how to work with the documentation.

cbrxyz, 2022-05-27:
This also needs to be fixed in Alabaster! Thanks for catching this Jarrod! 👍

cbrxyz, 2022-09-19:
This has been fixed. Thanks Jarrod!

closed 2022-09-19 by cbrxyz