uf-mil/mil #662: Create way of following an orange path marker

alexperez33, 2022-06-05:
Determine how to follow an orange path marker. This path marker will lead us to two of the tasks so we need to be able to make the sub follow the path that this marker is pointing towards and continue on that trajectory accurately until we hit the next task. 

OpenCV will likely be your friend for this task. 

In order to get started, you will likely want to check out the sub8_testing_2022 branch. This has some models in a world we have been playing with thus far. By launching ```roslaunch sub8_launch gazebo.launch```, you can launch the simulation of the sub. By then running ```gazebogui``` or ```rqt```, you should be able to see the models. (Note if you use rqt, you will have to open the camera plugins to see what the sub is seeing.

It would be wise to take images from rqt that you can use to develop your opencv code (when moving the sub over the path markers). This will allow quicker development. Feel free to move the path markers anywhere in the gazebo world for developmental purposes.

cbrxyz, 2022-06-14:
I'm interested in taking up this task. I'm going to start looking into how to at least find the general shape/color range of the path marker, without worrying about the possible effects of the light/pool/environment! I'll update when I have more information.

kawaiiPlat, 2022-08-01:
Nature of the event makes marker finding unnecessary, closing.

closed 2022-08-01 by kawaiiPlat