uf-mil/mil #680: Construct a physical start gate for the 2022 RoboSub competition

cbrxyz, 2022-06-12:
**This task is related to the 2022 RoboSub competition. Interested in helping out? Please click here to navigate to the [parent issue](https://github.com/uf-mil/mil/issues/608) or the [parent project](https://github.com/uf-mil/mil/projects/7), which can show you how to get started.**

This task requires the construction of a physical start gate for the 2022 RoboSub competition. The dimensions of the start gate and the materials used to make the start gate can be found on page 12 of the RoboSub 2022 rules manual (a link to the manual can be found on the parent issue linked above).

To complete, the task, construct a physical start gate that meets the appropriate requirements. You do not need to add any code to complete this issue.

A quick view of the start gate requirements can be seen below:
[image]

cbrxyz, 2022-06-12:
Currently, the physical start gate is not the appropriate color. That is all that should need to be fixed for this task to be marked as complete, for all that I am aware.

mitchellrobert, 2022-06-23:
Unless someone else made another start gate, the one we have is only 1in pvc while we need a gate that is 3in pvc

kawaiiPlat, 2022-07-23:
Start gate was assembled, closing

closed 2022-07-23 by kawaiiPlat