uf-mil/mil #682: Construct two physical Make the Grade buoys for the 2022 RoboSub competition

cbrxyz, 2022-06-12:
**This task is related to the 2022 RoboSub competition. Interested in helping out? Please click here to navigate to the [parent issue](https://github.com/uf-mil/mil/issues/608) or the [parent project](https://github.com/uf-mil/mil/projects/7), which can show you how to get started.**

This task requires the construction of the physical Make the Grade buoys for the 2022 RoboSub competition. The dimensions of the buoys and the materials used to make the buoys can be found on page 14 of the RoboSub 2022 rules manual (a link to the manual can be found on the parent issue linked above).

To complete, the task, construct two physical buoys that meet the appropriate requirements. You do not need to add any code to complete this issue.

A quick view of the start gate requirements can be seen below:
[image]

cbrxyz, 2022-06-12:
We have the image of the badge, but are still missing the image of the gun.

kawaiiPlat, 2022-07-23:
Bouys were assembled, closing

closed 2022-07-23 by kawaiiPlat