uf-mil/mil #758: Test issue for InvestiGator

cbrxyz, 2022-07-09:
Oh no! This issue has definitely been opened by someone who is not part of the uf-mil organization!

Sounds like someone needs to chomp on this one...

uf-mil-investigator, 2022-07-09:
@Object, thank you for your interested in the projects being worked on at the [Machine Intelligence Lab at the University of Florida](https://mil.ufl.edu)!

Unfortunately, at this time, we are unable to accept contributions from users outside of our organization for this repository. We thank you for your interest!

_If you are a member of MIL, please contact a software leader to get access to the @uf-mil organization._

uf-mil-investigator, 2022-07-09:
@octocat, thank you for your interested in the projects being worked on at the [Machine Intelligence Lab at the University of Florida](https://mil.ufl.edu)!

Unfortunately, at this time, we are unable to accept contributions from users outside of our organization for this repository. We thank you for your interest!

_If you are a member of MIL, please contact a software leader to get access to the @uf-mil organization._

uf-mil-investigator, 2022-07-09:
@cbrxyz, hola from the [Machine Intelligence Lab at the University of Florida](https://mil.ufl.edu)!

Unfortunately, at this time, we are unable to accept issues and contributions to this repository from users who are not yet part of MIL. We may accept contributions in the future, and some of our other repositories and projects accept contributions. However, because much of our lab is dependent on this repository and because we use several in-house tools to ensure that new additions work properly, we are unable to easily accept the contributions from researchers outside of MIL.

_If you are a member of MIL, please contact a software leader to get access to the @uf-mil organization._

closed 2022-07-10 by uf-mil-investigator