uf-mil/mil #810: broken link in vrx docs

kawaiiPlat, 2022-08-16:
[image]
In [this page](https://uf-mil.github.io/docs/navigator/vrx.html) there is a broken link in the warning at the top. to resolve, make this link an actual link and not just some text.

PrajwalBorkar, 2022-08-16:
I'm trying to work on this. But due to not having enough knowledge on .rst files,I'm unable to add the link for that directory. Any suggestions would be helpful.

uf-mil-investigator, 2022-08-16:
@PrajwalBorkar, hola from the [Machine Intelligence Lab at the University of Florida](https://mil.ufl.edu)!

  Unfortunately, at this time, we are unable to accept issues and contributions to this repository from users who are not yet part of MIL. We may accept contributions in the future, and some of our other repositories and projects accept contributions. However, because much of our lab is dependent on this repository and because we use several in-house tools to ensure that new additions work properly, we are unable to easily accept the contributions from researchers outside of MIL.

  _If you are a member of MIL, please contact a software leader to get access to the @uf-mil organization._

PR #824 (2022-08-18): Change link to RST format
Fixes #810.
closed 2022-08-18 by cbrxyz