uf-mil/mil #845: Fix is_simulation vs is_vrx redundancy

alexperez33, 2022-08-26:
In the ```navigator_launch``` package, you have ```simulation.launch```. This file launches the navigator simulation with most of the same code that will run on the actual robot. There are rosparameters established in this launch file called is_vrx and is_simulation. These are always true at the same time (from what I see), and they basically mean the same thing. Modify this file, and any affected files to combine these rosparamters into one rosparameter (you can choose to keep is_vrx or is_simulation as the param name). 

Also make sure along the way that doing the above process makes sense. And if it doesn't make sense, then we may need to reconsider keeping them separate.

Nicobetancur, 2022-08-27:
finished

cbrxyz, 2022-09-02:
Unfortunately, it was found after some investigation by @alexperez33 that this change is not needed as having both parameters is important to the proper functioning of the simulation.

closed 2022-09-02 by cbrxyz