uf-mil/mil #858: Tool for activating thrusters

BlakeSanders10497, 2022-08-31:
We want an alias or program that can choose a thruster and spin it in one simple command.

Original command: `rostopic pub /BR_motor/cmd roboteq_msgs/Command "mode: 0 setpoint: 0.0"`

To be replaced with something similar to: `thruster BR 0.0`

Okayd123, 2022-09-08:
I noticed that this already exists in `NaviGator/scripts/bash_alias.sh` as `nthrust`. 
Not sure if anything else needs to be added to it, but yeah.

cbrxyz, 2022-09-08:
Thanks for noticing that Henry! We should add some documentation on this tool and ensure that it works on Sunday.

cbrxyz, 2022-09-19:
We are planning to add this to our new NaviGator focused-CLI, `nav`. If any members are still interested in working on it, please let me know! Thank you!