uf-mil/mil #893: Install SSH server on Zobelisk

cbrxyz, 2022-09-14:
### What needs to change?

Zobelisk, one of our main lab computers, needs an SSH server so that lab members can access the computer without needing to physically be next to Zobelisk.

This SSH server would be restricted to the MIL network.

### How would this task be tested?

1. Create an SSH server on Zobelisk.
2. Sign into the SSH server while on the MIL network. You should simply be able to use `ssh zobelisk@ip` to access the computer.

### Contacts

- [ ] We need help from the mechanical team.
- [ ] We need help from the electrical team.
- [ ] We need help from Dr. Schwartz or other faculty.
- [ ] We need help from a company or an organization.
- [ ] We need help from another UF staff member or organization (ex, facilities).

tringuyenswe, 2022-10-04:
It works. Can test using "ssh zobelisk@192.168.37.176" while on MIL network.

closed 2022-10-04 by tringuyenswe

cbrxyz, 2022-10-05:
Thanks for your help with this Tri!