uf-mil/mil #907: `mil` Unified Command Project

cbrxyz, 2022-09-24:
<h1 align="center"><code>mil</code> Unified Command Project</h1>

The `mil` unified command project is a project to unite the various aliases and commands used commonly by MILers to perform actions. Currently, these commands and aliases are scattered throughout the repository, which sets up a challenge for old and new members looking to complete software tasks. Furthermore, documentation can be hard to find or nonexistent for these commands, which provides an even greater barrier to entry.

To solve this, we are starting this project, which will aim to unite all of our commands into one program: `mil`. For example:
* `nmove` (move NaviGator in a direction) → `mil nav move`
* `subviz` (open SubjuGator Rviz config) → `mil sub rviz`
* `list_mil_devices` (list all devices on MIL's network) → `mil network listdevices`

Furthermore, members will be able to run `--help` on the `mil` command to find out more about what the command (and each sub command) does:
```bash
$ mil --help

mil [COMMAND] [ARGS]

 nav           Manipulate NaviGator and its functioning
 sub           Manipulate SubjuGator and its functioning
 network       View and modify info related to the MIL network
 ...

$ mil nav --help

mil nav [COMMAND] [ARGS]

 control       Control NaviGator
 thrust        Spin NaviGator's thrusters
 rviz          Open NaviGator model in Rviz
```

This project and each command will be implemented in Python to make the project code easier to read and implement. The tracking branch for the project will be `nav-cli` - members can branch off of this branch and make PRs back into it as needed.

<hr>

**How do you get started/get involved?** Just take on one of the tasks listed in the table below. Each task is focused on one individual command, for example, `mil nav control xbox`. For more of an intro on how to use Git, please see our [contributing guide](https://uf-mil.github.io/docs/software/contributing.html).

tringuyenswe, 2023-01-03:
Hi! I'm not sure if this has been worked on, but I was able to come up with multi-line bash commands, which I think is really useful and I don't see much good implementation online. I think I will be too busy next semester and can't work on anything for MIL, but I hope this is helpful for anyone trying to tackle this project. One thing to keep in mind is to remove all existing bash alias for like mil or nav or else this won't work and will just redirect to the old alias. Let me know if there are any questions : ) 

[multiline_bash_example.txt](https://github.com/uf-mil/mil/files/10340315/multiline_bash_example.txt)

cbrxyz, 2023-02-09:
This issue is being closed until the repository is migrated to ROS 2 and is therefore compatible with Rust CLI tools.

closed 2023-02-09 by cbrxyz