uf-mil/mil #1038: Remove the `deprecated` folder from the repository

cbrxyz, 2023-04-12:
### What needs to change?

Our repository contains a `deprecated` folder for storing code that members want to keep around, but is no longer used. Rather than storing this code inside a `deprecated` folder, we should move this code into our documentation site, for several reasons:
* Confusion amongst new members ("Why is there a deprecated folder?")
* Needing to exclude search results from `deprecated` when attempting to find resources using tools like `fd` or `grep`
* Moving the code into documentation can make it easier to find for new members and those not on a computer

This task requires going through the code listed in `deprecated` and moving the code into our documentation. This will require sifting through the code, determining what is important, and then putting that into a cute format for our documentation! :)

### How would this task be tested?

1. This task is simply code cleanup and documentation, so there shouldn't need to be any tests. Just ensure that nothing is broken by moving around different files and pieces of files!

### Contacts

- [ ] We need help from the mechanical team.
- [ ] We need help from the electrical team.
- [ ] We need help from Dr. Schwartz or other faculty.
- [ ] We need help from a company or an organization.
- [ ] We need help from another UF staff member or organization (ex, facilities).

MrinallU, 2023-09-06:
On second thought I'll probably come back to this once I have more experience.

kawaiiPlat, 2023-09-08:
Checking if bot works