uf-mil/mil #1140: Convert Trained YOLO Model to Tensorflow Lite

andrew-aj, 2024-02-28:
### What needs to change?

As part of our move to Sub9, we need to get our vision models compatible with the computer. Traditionally, running YOLO requires a GPU which we have on Sub8 to run our ML pipeline on. However, Sub9 is a smaller form factor and as a result we cannot fit a GPU. Running our vision models just on the CPU would be too intensive but we have a potential work around. We bought an M.2 TPU accelerator [here](https://coral.ai/products/m2-accelerator-dual-edgetpu). However, it only runs TensorFlow Lite whereas our current YOLO model uses PyTorch. There is a potential work around where we can convert PyTorch models to TensorFlow Lite. This task is to determine if this is possible. Here is a link that you can use to start from: [converting models](https://medium.com/@zergtant/convert-pytorch-model-to-tf-lite-with-onnx-tf-232a3894657c).

### How would this task be tested?

1.
2.
3.

### Contacts

- [ ] We need help from the mechanical team.
- [ ] We need help from the electrical team.
- [ ] We need help from Dr. Schwartz or other faculty.
- [ ] We need help from a company or an organization.
- [ ] We need help from another UF staff member or organization (ex, facilities).

DaniParr, 2024-03-05:
I was able to successfully make the tflite model, I plan to make a submodule for our repo to convert any .pt model into tflite model!

cbrxyz, 2024-07-17:
This is only relevant to sub9, not for sub8, so I'm going to close it.

closed 2024-07-17 by cbrxyz