uf-mil/mil #1244: Setup ROS2 bridge for VRX

cbrxyz, 2024-08-15:
### What needs to change?

To use the new version of the VRX simulator (ie, the simulator for RobotX), we will need to install the ROS 2 bridge, which will allow our ROS 1 code to communicate with the ROS 2 code used by VRX.

### How would this task be tested?

1. Add a test to ensure that we can use the VRX simulator (ie, we can run a launch file and launch our boat).

tjcanro, 2024-09-16:
### Update:

I followed installation steps for my local ubuntu 20.0.4 machine for the VRX simulator. This involved building a docker image as specified in the VRX repo. I also read the documentation on the ros1_bridge package and its usage, as well as multiple bridge tutorials on the different options this package and other similar ones offer. At the moment, I have the following preliminary steps in mind for the bridge setup:

- Understand the requirements for ROS1 communication with VRX ROS2 code.

- Meet with team leaders (@cbrxyz) to clarify environment-specific matters in our project.

- Setup docker image for virtual machine (as done locally) or ROS2 distribution.

- Install the ros1_bridge package. Either for one-way or dynamic communication between ROS1 and ROS2 topics.

- Verify basic connectivity and functionality of the bridge by creating a test nodes .

- Modify launch files that ensure the proper bridge configuration and setup.

- Launch and verify functionality of the VRX simulator.

- Create test case that automatically tests the bridge upon launch.

tjcanro, 2024-09-23:
### Progress update: 

I am working on the installation of ros2, which is a prerequisite for ros1_bridge. I am writing a script that handles this installation, so that it can be ran before running the new VRX version.

I have read documentation and guides for the use of the bridge, and knowing that it needs two-way communication, a dynamic bridge is likely required. 

In order to ensure the bridge is working properly, I will be writing a test node and a guide for testing the functionality of the bridge.

Finally, I had some issues with the initial approach I was considering, as it filled my Linux disk space, so I had to handle that before I could continue with the bridge setup

tjcanro, 2024-10-07:
### Progress update:

This week I did not manage to work on this issue. I had interviews scheduled on a very short time frame throughout the week so I will make up for the time next week.

tjcanro, 2024-10-14:
### Progress Update:

I managed to get the bridge running on my local machine using foxy as my ros2 distro. I am working on adding its installation to the scripts. Before submitting a PR I'll have to test my changes to ensure they don't break our environment

tjcanro, 2024-11-11:
### Update

- I have completed the bridge installation. Although I had the PR setup for ROS2 foxy, due to incompatibility with the new gazebo garden that must be used for VRX, I had to modify it to work with ros2 humble
- I am drafting the PR for the humble/noetic bridge. 
- Aside from drafting the PR. The following package must be fixed as well: 'rcl_interfaces/msg/Log' and ROS 1 type 'rosgraph_msgs/Log'