uf-mil/mil #1254: RobotX 2024: Task 3 – Follow the Path

DaniParr, 2024-09-01:
### What needs to change?

We need to make sure that the current [mission](https://github.com/uf-mil/mil/blob/master/NaviGator/mission_control/navigator_missions/navigator_missions/navigation.py) works correctly and the WAMV can navigate through all reasonable path configurations.

[image]

### How would this task be tested?

1. Launch ROS with your versions of gymkhana like such: `roslaunch navigator_launch simulation.launch world:=2022_practice/gymkhana0 --screen`
2. Launch the mission: `mission run Navigation`

EAOZONE, 2024-10-06:
Don't know if I will be of any help on this task. However, this week, I started looking at how the old navigation course code was written and making sure I understood the ins and outs of that.

thadavale, 2024-10-14:
This week I began looking through the navigation code and understanding how it works in simulations by testing it on practice courses in Gazebo.

EAOZONE, 2024-10-14:
This week I went home for the hurricane, so didn't get much done. However, I did take some time to understand more of the previous code.

thadavale, 2024-10-27:
This week, I tested the navigation code in Gazebo on the previously written test cases for Task 3.

thadavale, 2024-11-04:
I continued testing the navigation code in Gazebo on the previously written test cases but I noticed a few issues. Firstly, the test cases have white cones at the beginning and end of the courses, which the navigation code tends to get confused by. These are not present in the actual competition, so they should be removed. The actual competition has a single multicolored marker at the beginning and end of the course that may need to be accounted for in the navigation code. Additionally, I noticed that the wamv keeps investigating the obstacles but often is not able to figure out how to travel through them. Also, I kept receiving an error that my gazebo process died while the Navigation code was running, likely from a segmentation fault, so I will try to figure out how to fix this error.