uf-mil/mil #1256: RobotX 2024: Task 5 - Scan the Code

DaniParr, 2024-09-01:
### What needs to change?

For this task, the WAMV needs to first find where the three-light sequence display is and then record the RGB sequence it observes.

[image]

### How would this task be tested?

1. Launch ROS for navigator using the world as described in this command line: `roslaunch navigator_launch simulation.launch world:=scan_dock_deliver --screen`
2. Then run the mission: `mission run ScanTheCodeMission`

DaniParr, 2024-09-09:
@mohana-pamidi @hoshbosh @edwardguthrie Any progress on this task or the approach to it?

hoshbosh, 2024-09-10:
@DaniParr @mohana-pamidi  @edwardguthrie  I am looking through the handbook right now and this is what I have so far for the scanning part of the task. 
[image]
For reference, the color memory is just an array of length three that gets filled with the data of the sequence. I was thinking this loop could run for each frame passed in from the camera, after the WAMV has already navigated to the light display.

hoshbosh, 2024-09-23:
I started to write a basic mission for the task, just moving around identifying the light display. That is still what I plan on working on next week.

EAOZONE, 2024-10-20:
This week worked with Keith to figure out what needs to be changed for the task from past code. Including: adding the computer vision model etc.

EAOZONE, 2024-10-27:
This week I looked at how all the different tasks were running as of now and took notes on why I thought that was the case, looking inside the olde code and running the simulation.

EAOZONE, 2024-11-04:
This week I got the computer vision detection working for the scan the code task. Making it be able to get the correct sequence.