uf-mil/mil #1257: RobotX 2024: Dock and Deliver

DaniParr, 2024-09-01:
### What needs to change?

Write a mission where the WAMV is given a color (red, green, or blue) and it must dock into the right colored dock. Then the WAMV must shoot balls into one of the black holes.

[image]

### How would this task be tested?

1. Launch ROS with this world: `roslaunch navigator_launch simulation.launch world:=scan_dock_deliver --screen`
2. We will then run either your mission or the mission you modified: `mission run <mission_class_name>`

aaliao, 2024-09-08:
[image]

DaniParr, 2024-09-09:
@aaliao Great workflow diagram! I really liked how you highlighted which components of the mission are left to be completed! We already have an issue for "Detect where to Dock" that is #1261. I suggest focusing on the portion of the last two segments of your diagram!

andrew-aj, 2024-09-13:
In docking_2024 the current code in docking.py correctly moves to the dock and docks in the left most docking bay. Others are currently working on finding the color of the dock to figure out which bay to dock in.

aaliao, 2024-09-15:
I have been working with the simulation and looking through the code to get a feel for how the missions code are structured- for aligning the boat with the holes I have been talking to @danushsingla about using a similar code for aligning with center of colored pixel clusters to move the boat in position, or possibly aligning it with the hole with CV, similar to what he is doing for the detect to dock code. There also exists a function for shooting, however in the sim it doesn't seem to be implemented, however there is a function to call in the base.

AlexanderWangY, 2024-09-22:
I have been contributing to a color detection python script to try and mask out squares as well as detect the color of a certain pixel. This week we have decided to move forward with a color threshold tactic to decipher whether the pixel is red, green, or blue. Although masking could work, if all goes well and the LIDAR correctly crops out the boards then the center pixel cluster should accurately represent the color of the board. Therefore we will simply extract the center pixels and average their pixel intensities in the RGB scale. Then we will be able to send back the detected color.

Moving forward, I am going to modularize the pixel averaging and color detector module into a class that can be imported into `Docking.py` for further use by the missions team.

aaliao, 2024-09-22:
I worked with @AlexanderWangY and @danushsingla and @willzoo with the earlier section for detecting to dock. Alex has implemented code to mask the square colored circles, and I will be implementing that this week to get the center point of the square within the image to align the boat for shooting. Last week I looked a bit with the shooting function in the simulation and something of how I implemented it seemed wrong, so I will try investigate that as well, however it seems to be a simple await call for shooting and reloading implemented in navigator.py

aaliao, 2024-09-30:
Status update:
At the same place since the last update, was not able to get much progress at all with the hurricane

Carlosdc25, 2024-10-14:
This week I learned to set up the simulation and run commands for the simulated navigator. I then started working on task 6 with @aalio. For the most part, I learned how the already-established code worked and what problems were left to solve.  Particularly, I learned about the issue of centering the boat to one of the holes in which to shoot the ball. The method we are trying to use involves using contours obtained from the camera once the boat is docked in front of the holes, and then moving the boat in relation to the image it sees. Exactly how the boat needs to be moved is going to depend on where the shooter is on the boat.

aaliao, 2024-10-14:
I added code in the navigator_vision module to isolate the colored square after docking to align the boat with the hole before shooting. It hasn't been fully implemented into docking.py yet however, and I spent time trying to get it running in the simulator however I haven't got it fully working in the simulation. Using example images from the simulation and generated pictures, it seems to work well in finding the center square to align with, however this needs to be tested in simulation and with varying weather conditions outside to test what bounds the CV transformations should be set to.

[image]
[image]
This image shows the part of the pipeline to isolate the square. Following this, we should be able to get the center point or other values from its contour.

aaliao, 2024-11-18:
Nothing done this week post competition, I assume this issue will be closed soon.