uf-mil/mil #1262: RobotX 2024: Task 3 - Follow the Path TEST CASES

DaniParr, 2024-09-02:
### What needs to change?

Write test cases for [Task 3](https://github.com/uf-mil/mil/issues/1254). Create these by modifying existing worlds and saving those world files to be deployed on ROS launch as an argument.

### How would this task be tested?

1. Launch ROS: `roslaunch navigator_launch simulation.launch world:=<name_of_world> --screen`

thadavale, 2024-09-08:
I created a branch that includes test world files for this issue inside the worlds folder.

thadavale, 2024-09-09:
For this task, I went over the basics of using Gazebo. The world file I created is named robotx_task_3_test_cases.world and has an example path for the robot to go through with red and green buoys. 
[image]

DaniParr, 2024-09-09:
@thadavale Amazing job! Lets get a few more worlds to test out edge cases but amazing job either way!

@EAOZONE @tjcanro Please get in contact with Tanushree on Discord and help her create more worlds to test the Navigation task!

EAOZONE, 2024-09-13:
This week, using a pre built model, I made a 2 buoy system so that we could design different test cases such as the one given for robotx. I then designed the given test case so that we can start making more by editing the poses of the 6 buoy models.

EAOZONE, 2024-09-14:
Created two tests cases one facing the left one facing the right. From here should be pretty easy to change the two and make loads of different test cases for Task 3
[image]
[image]

thadavale, 2024-09-16:
This week, I created two new test world that include the white buoys at the start and end of the course as well as the round black obstacle buoys. These test paths are in U-shapes going in both directions. I also created a path with a sharp right turn.
[image]
[image]
[image]

EAOZONE, 2024-09-22:
This week I made 2 more test cases for the task, one as a zigzag and one a hard left turn

thadavale, 2024-09-23:
This week, I created 3 new test cases. One of them is a straight path and the other two are L-shaped paths going both left and right. I also updated the paths all of the test worlds to work on any user's computer. 
[image]
[image]
[image]

EAOZONE, 2024-09-29:
This week I fixed the problems that @cbrxyz found in our pull requests for both Task 3 test cases and task 4 test cases. This included removing the Wamv model from every test case as well as changing the name of the different models created in task 4