uf-mil/mil #1263: RobotX 2024: Task 4 - Wildlife Encounter TEST CASES

DaniParr, 2024-09-02:
### What needs to change?

Write test cases for [Task 4](https://github.com/uf-mil/mil/issues/1255). Create these by modifying existing worlds and saving those world files to be deployed on ROS launch as an argument.

### How would this task be tested?

1. Launch ROS: `roslaunch navigator_launch simulation.launch world:=<name_of_world> --screen`

DaniParr, 2024-09-09:
@MarcinPlaza @EAOZONE Hey y'all any progress on this issue?

EAOZONE, 2024-09-09:
> @MarcinPlaza @EAOZONE Hey y'all any progress on this issue?

I'm new and was just figuring things out last week. Plan on working on it this week.

MarcinPlaza, 2024-09-13:
[image]
Well I wasted some time trying to make better models, but I ended up just making basic models for the buoys. Here's my understanding of the positioning. I will push to a branch soon. Should I make separate worlds for the different cases randomizing the color order?

EAOZONE, 2024-09-21:
This week I took the created test case by @MarcinPlaza and made 5 other versions of it with the different color order

hoshbosh, 2024-10-07:
This week I went through the pr by @DaniParr on my local machine, loading the simulation and reviewing the code.

PR #1285 (2024-11-09): Wildlife mission
## Description

Adapted the wildlife mission with the simulated props (RGB buoys) and a corresponding world (wildlife_20m_triangle_no_timer.world).

## Screenshot or Video

Video is on Zobelisk => Videos/wildlife2024.mp4

## Related Issues

\- Closes #1263

## Testing

1. `roslaunch navigator_launch simulation.launch use_mil_world:=true world:=wildlife/wildlife_20m_triangle_no_timer --screen`
2. Drag boat to the middle of the three buoy props so lidar can pick up on buoys
3. Label the three buoys accordingly (red_python_buoy, blue_manatee_buoy, green_iguana_buoy)
4. `mission r [... cut]
closed 2024-11-09 by cbrxyz