uf-mil/mil #1264: RobotX 2024: Task 5 - Scan the Code TEST CASES

DaniParr, 2024-09-02:
### What needs to change?

Write test cases for [Task 5](https://github.com/uf-mil/mil/issues/1256). Create these by modifying existing worlds and saving those world files to be deployed on ROS launch as an argument.

### How would this task be tested?

1. Launch ROS: `roslaunch navigator_launch simulation.launch world:=<name_of_world> --screen`

DaniParr, 2024-09-09:
@hoshbosh @edwardguthrie Hey y'all any progress on this issue?

hoshbosh, 2024-09-09:
@DaniParr Yeah I was able to make some worlds with the display in different places relative to the boat, but I was having some trouble getting it pushed. I was planning on coming into the lab to get that sorted out.

DaniParr, 2024-09-09:
@hoshbosh Sounds good. I will be in the lab during my office hours and this week Wednesday from 1:00 -> 4:00 PM.