uf-mil/mil #1268: Computer vision for drone landing pags

cbrxyz, 2024-09-04:
### What needs to change?

We will need to develop a computer vision model to detect the drone landing pads. @LesterBonilla found a camera that does automatic computer vision from some self-developed machine learning model, which we will try first. If that camera's built-in vision model does not perform at the level we desire, then we can develop our own ML vision model that will run directly on the drone's pi.

### How would this task be tested?

1. Ensure that the drone landing pad can be detected in a variety of environmental conditions, with the tins on it and not.

cbrxyz, 2024-09-10:
Hey Josh, I saw you were assigned to #1267 as well - can you just focus on this task for now?

Josht8601, 2024-09-10:
Hey Cameron! Yeah, I can focus on that task for now.

maxhv1, 2024-10-20:
Status report: I attended the WAMV test on 10/16/24. I'll begin working on this issue soon. The plan for this, as suggested by @Keith-Khadar , is to create the landing pads in gazebo and develop a machine learning model based on that. I have the needed landing pad logos from RoboNation and the landing pad dimensions from the manual. I will make a set of images of the landing pad gazebo models, then label them in label studio, and then train a MIL Yolo model off of those images.

maxhv1, 2024-10-28:
Status report: 
[image]
I made the landing pads and cones in Blender, exported them for use in gazebo. I also labelled some images this week for small buoy detection for the animals task. I think I need to either make a new ros package and/or launch file for gazebo, since I think we want to run this gazebo world without the WAMV and try Gazebo with SITL for simulating the drone. I had tried running navigator_launch simulation.launch on a world file with the WAMV removed, it didn't go well. I tried installing Ardupilot earlier in the week also, but I mixed up which installation process to use (we need the legacy one for Ros1), haven't got that working yet. I think I messed up my gazebo doing that, I had to reinstall gazebo and now it looks like 
[image]

maxhv1, 2024-11-04:
Status report: I was unable to complete the gazebo world in time. I did help the mechanical team make a stencil for the Robonation objects for Task 8 (relevant to this issue), but I did not get to make a data set nor machine vision model off of it unfortunately. I tried to help figure out the params.py for path planner, but due to using a VM I was unable to run Gazebo and Nviz effectively.