uf-mil/mil #1274: Add communication library for new Light Tower Communication Board

cbrxyz, 2024-09-08:
### What needs to change?

We will need a library to work with the Light Tower board being developed by our electrical team (see uf-mil-electrical/NaviGator#2). This would include telling the board when we have switched between manual and autonomous control (as it should be able to derive when we are killed from the kill/relay board).

It looks like the inputs will be three high/low signals, so we should just be able to control the high/low statuses of the signals we are sending to the board to control it, but we will see!

### How would this task be tested?

1. Ensure that our library is able to communicate with the new board.
2. Ensure that our library can't accidentally output two states at once.
3. Ensure that our library correctly responds to switching between autonomous/manual modes.

cbrxyz, 2024-09-08:
@tjcanro Do you want to try taking this on? It shouldn't be too much. You will need to work with Adrian on the specifics of how the board communicates with our library/the computer, but it shouldn't be overly complex.

tjcanro, 2024-09-08:
Yeah for sure!

cbrxyz, 2024-09-19:
I misunderstood the electrical configuration: we won't be talking to the light tower board directly; rather we will talk to the RF Kill Board which will then talk to the light tower for us.

Shxl2, 2024-09-30:
After coming to testing today, I decided that I want to try to make another communication library for the robot. I haven't been able to do anything yet because of the hurricane and career fair, but will begin working on this task this week.

cbrxyz, 2024-09-30:
@Shxl2 Hey, this one probably also isn't the best to do, sorry. I'm about 50% done with the driver so far. Really sorry about this, we just have to complete it soon since it is required for us to have to compete.