uf-mil/mil #1275: Add communication library for ball shooter board

cbrxyz, 2024-09-08:
### What needs to change?

We will need to design a library to speak with the new ball shooter/actuator board. This board will likely include two controls:

1. Spin up the motors quickly
2. Drop a ball into the motors

We will need to a design library that can communicate with this board. The design (ie, ROS topics/services/etc.) should mirror the method to fire balls in VRX.

Relevant electrical issue: uf-mil-electrical/NaviGator#3

### How would this task be tested?

1. Ensure that our library successfully communicates with the board/device.
2. Ensure that the ROS inputs are visible/can be called.
3. Ensure that no improper states can be set. (ie, will trying to spin up the motors once they are already spinning cause an error?)
4. Ensure that the ROS inputs trigger the board properly

Shxl2, 2024-09-14:
[image]

I'm planning to make a ROS node similar to the HW kill driver that will handle all the codes that are sent and received to the board and will subscribe to topics that control the motor and servo from other nodes. Ideally the node would also have publishers that output telemetry from the actuators but I do not yet know the exact details of what is being communicated from the board. So far in the flowchart I just have the logic for the connection to the board through serial.

Shxl2, 2024-09-23:
For this week, I have begun writing code for the ROS package. Right now, I am copying the code for the HW kill driver and using that as a template for the actual driver and it's sim classes. When I receive the codes needed to communicate with the board, I will add them to the constants file, and write the methods to shoot. Until then, the methods will be empty.

cbrxyz, 2024-09-27:
@Shxl2 - Hey Javier! I hope you're doing well, and staying safe. Super sorry, but I accidentally completed this issue and didn't realize that you were assigned to it/interested in doing it. I would reach out to Daniel (your mentor) to find another task that you can work on. You're welcome to reach out to me as well! Again, super sorry about this!

commit 4cd65d2f (2024-09-27): navigator_ball_launcher: Add test cases, add service for retrieving number of balls dropped, move driver to Python library, fix packet class checker when using subclassed generics in electrical_protocol (closes #1275)
closed 2024-09-27 by uf-mil-bot