uf-mil/mil #1281: Migrate simulated and world camera topic names to consistent format

cbrxyz, 2024-09-25:
### What needs to change?

Right now, the simulated camera topics show up as:
```
/wamv/sensors/cameras/front/left/image_raw
```

The real-life camera topics show up as:
```
/camera/front/left/image_raw
```

We should migrate these topic names (and nodes who use them) to a consistent format!

### How would this task be tested?

1. Ensure that all tests pass and NaviGator doesn't break once these changes are made!

thadavale, 2024-09-29:
This week, I created a branch for this topic and began looking into the topic names and the nodes that use them. I also began compiling a list of files that use the relevant topic names and researching how to migrate topic names.

thadavale, 2024-10-07:
This week I updated all of the rostopic names to be consistent with each other.

lynettehemingway, 2024-10-14:
This week I was unable to work on as much as I would like due to the hurricane. However, I learned how to migrate topic names and plan on helping fix the camera info error.

thadavale, 2024-10-21:
This week, I tried many different methods and was finally able to fix the error I previously encountered by remapping the old simulation-specific topic names onto the new real-life topic names inside simulation.launch and vrx_classifier.launch. I also made a pull request.